langwadt/grbl_stm32 #1: Pin compatability

VojislavM, 2016-03-30:
Hi @langwadt 

is there any particular reason why pins in your version of stm32-grbl on [NUCLEO-F411RE](http://www2.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-f411re.html) are not compatible with [Arduino Uno pins with GRBL firmware](https://github.com/grbl/grbl/wiki/Connecting-Grbl)?

langwadt, 2016-03-30:
yes, I'm using it with a stack of three of these: http://www.st.com/web/en/catalog/tools/PF260715
they have their own pinout and need pins for spi to setup the stepper drivers

it should be easy to change to a different pinout by just redefining the pins in the cpu_map  file, 
the only pin that can't just be moved is the spindle control because that is on a timer pin

VojislavM, 2016-03-31:
I changed pin positions in the code, but it is still not working, I moved SPI1 to different SPI pins because they were overlapping with Arduino header pins. I saw you changed parts of the code in stepper.c, and in some other files. 

Can you please share your opinion what else should I change to get Nucleo running with Arduino stepper sheld?

```
#define X_STEP_BIT      0  // Uno Digital Pin 2
#define Y_STEP_BIT      1  // Uno Digital Pin 3
#define Z_STEP_BIT      2  // Uno Digital Pin 4
#define STEP_MASK       ((1<<X_STEP_BIT)|(1<<Y_STEP_BIT)|(1<<Z_STEP_BIT)) // All step bits

#define X_DIRECTION_BIT   3  // Uno Digital Pin 5
#define Y_DIRECTION_BIT   4  // Uno Digital Pin 6
#define Z_DIRECTION_BIT   5  // Uno Digital Pin 7

#define X_LIMIT_BIT      0  // Uno Digital Pin 9
#define Y_LIMIT_BIT      1  // Uno Digital Pin 10
#define Z_LIMIT_BIT	     2 // Uno Digital Pin 12

#define RESET_BIT         0  // Uno Analog Pin 0
#define FEED_HOLD_BIT     1  // Uno Analog Pin 1
#define CYCLE_START_BIT   2  // Uno Analog Pin 2
#define SAFETY_DOOR_BIT   3  // Uno Analog Pin 1 NOTE: Safety door is shared with feed hold. Enabled by config define.
#define CONTROL_MASK      ((1<<RESET_BIT)|(1<<FEED_HOLD_BIT)|(1<<CYCLE_START_BIT)|(1<<SAFETY_DOOR_BIT))
#define CONTROL_INVERT_MASK   CONTROL_MASK // May be re-defined to only invert certain control pins.

#define PROBE_BIT       0  // Uno Analog Pin 5
#define PROBE_MASK      (1<<PROBE_BIT)

#define DIRZ     GPIOA,GPIO_Pin_8//
#define DIRY     GPIOB,GPIO_Pin_10//
#define DIRX     GPIOB,GPIO_Pin_4//

#define STPZ     GPIOB,GPIO_Pin_5
#define STPY     GPIOB,GPIO_Pin_3 //
#define STPX     GPIOA,GPIO_Pin_10

#define SPI1SCK GPIOB,GPIO_Pin_10//SPI2SCK
#define SPI1MOSI GPIOC,GPIO_Pin_3//SPI2MOSI
#define SPI1MISO GPIOC,GPIO_Pin_2//SPI2MISO
#define SPI1NSS GPIOB,GPIO_Pin_12//SPI2NSS

#define STP_RST GPIOA,GPIO_Pin_9
#define STP_FLG GPIOB,GPIO_Pin_13

#define TESTP GPIOB,GPIO_Pin_8

#define LIMX GPIOC,GPIO_Pin_7//GPIOA,GPIO_Pin_0
#define LIMY GPIOB,GPIO_Pin_6//GPIOA,GPIO_Pin_1
#define LIMZ GPIOA,GPIO_Pin_7//GPIOA,GPIO_Pin_4

#define PROBE GPIOC,GPIO_Pin_0//GPIOC,GPIO_Pin_1

#ifndef VARIABLE_SPINDLE
#define SPINDLE_EN   GPIOA,GPIO_Pin_6//GPIOC,GPIO_Pin_1
#define SPINDLE_DIR  GPIOA,GPIO_Pin_5//GPIOC,GPIO_Pin_1
#endif
//#define FLOOD_COOLANT
//#define MIST_COOLANT

//#define RESET_PIN
//#define FEED_HOLD_PIN
//#define CYCLE_START_PIN
//#define SAFETY_DOOR_PIN

#define XCURRENT 1500
#define YCURRENT 1500
#define ZCURRENT  500
```

VojislavM, 2016-03-31:
I think maybe the problem is in this three functions:
```
void config_steppers()
void enable_steppers()
void disable_steppers()
```
You use SPI to config, enable and disable stepper motors.

langwadt, 2016-03-31:
I've added the option to use standard GRBL pinout enabled with  #define STANDARD_GRBL in cpu_map_stm32f411_nucleos.h

I haven't got any stepper hardware to test it on but the right pins seem to be toggling

VojislavM, 2016-04-01:
Yes, it worked for me too yesterday before your modifications. I saw your commit, It is great. 
I will test it on corexy machine, write some documentation, and make pull request, so you can add that to your repo, if you like.

Cheers

Andypos, 2016-04-01:
Hi, @langwadt and @VojislavM 
Thanks a lot for last change . I can use stepstick ,now . I build small corexy machine to.
Regards

Andypos, 2016-04-02:
Hi,
According to my opinion processor runs at a clock HSI. To change to MCO must uncheck  #define USE_HSE_BYPASS and change HSE_VALUE    ((uint32_t)8000000)  ( 25000000 IS NOW).
(I do not have X3 crystal on board)
Regards

langwadt, 2016-04-03:
@VojislavM good to hear it works, I already made the changes for the pinout but let me know if you have other things
@Andypos You are correct, I've now change code/project to use the HSE, it should be a bit more accurate but not change anything else

Andypos, 2016-04-03:
@langwadt 
`/*!< Uncomment the following line if you need to clock the STM32F411xE by HSE Bypass
     through STLINK MCO pin of STM32F103 microcontroller. The frequency cannot be changed
     and is fixed at 8 MHz. 
     Hardware configuration needed for Nucleo Board:
     – SB54, SB55 OFF
     – R35 removed
     – SB16, SB50 ON */
 #define USE_HSE_BYPASS

#if defined(USE_HSE_BYPASS)     
#define HSE_BYPASS_INPUT_FREQUENCY   8000000
#endif /* USE_HSE_BYPASS */    
#endif /* STM32F411xE */
`

langwadt, 2016-04-03:
@Andypos I have changed to the code and build symbols to use HSE. Just uncommenting the line as the comment suggest isn't enough, the uart setup gets the baudrate wrong because it calls that uses  25MHz as the default HSE frequency

Andypos, 2016-04-03:
HSE_VALUE is define in line 128 stm32fxx.h
I change it for next version for example with X3 crystal
`  #define HSE_VALUE    ((uint32_t)8000000) /*25000000!< Value of the External oscillator in Hz */`

For check FPU  and SystemClock debug small add code:
int main(void)
{
#if 0
float a=1.5,b=1.5,c=0.0;
unsigned int clock_1;
 c=a*b;
 SystemCoreClockUpdate();
 clock_1=SystemCoreClock;
#endif
``
My  compile with -O2 version have only 38.2 kB and work stability.

langwadt, 2016-04-03:
@Andypos instead of changing stm32fxx.h, I just add HSE_VALUE=8000000 to the build symbols

Andypos, 2016-04-03:
@langwadt  OK. but set #define __FPU_USED 1

langwadt, 2016-04-04:
@Andypos  __FPU_USED it is set automatically when you set the floating point ABI to softfp or hard

Andypos, 2016-04-04:
@langwadt OK. Today I install new version STM WB IDE . ( normaly I use old version Coocox Eclipse ) 
All is  OK..Thanks.
@VojislavM  I delete from GRBL shield    5V pin and connect to this pad  3.3V
Not all input pin from nucleo accept 5V ! Stepstick work ok with 3.3V.and no problem with imput pin.
Raegards

Andypos, 2016-04-06:
@langwadt  Thanks for last change.I have a question. Why function:  void spindle_stop()  PWM is set to 1000 ? Must be 0 !?

langwadt, 2016-04-06:
@Andypos I'm using the spindle output to control a brushless spindle driven by an RC speed controller it needs 1m-1.5ms pulses. I'll add the option to do normal 0-100% pwm

Andypos, 2016-04-20:
@langwadt  
My proposal changes TIM3 for STANDARD_GRBL and PWM_SPINDLE in spindle_init():
TIM_TimeBaseStructure.TIM_Prescaler =100-1;  // 100 MHz /100=1MHz
  TIM_TimeBaseStructure.TIM_Period = 125-1;  // 1 MHz / 125 = 8000khz (125 us)
TIM_OCInitStructure.TIM_Pulse = 0;

J-Dunn, 2016-04-24:
@langwadt

thanks for putting this up. 

I'm a bit confused. I've just grabbed a copy of https://codeload.github.com/langwadt/grbl_stm32/zip/master

README.md says your test rig was using Gecko drivers. Is that comment obsolete now? 

Also re.  #define STANDARD_GRBL in cpu_map_stm32f411_nucleos.h

I find neither that #2 define nor that file.  Have I grabbed the wrong code ? 

Thanks.