langwadt/grbl_stm32 #2: Homing problem with corexy

VojislavM, 2016-04-05:
@Andypos you said that you have corexy machine, controlled with Nucleo board. 
Do you have problem with homing ($H)?

Mine just goes to +x limit switch, then goes back, and goes to +y but never touch switch.
I saw that GRBL head problem with homing of corexy, but I think they fix it.

VojislavM, 2016-04-05:
Now it works, I have changed some things in  **src/grbl/limits.c**
You can see the changes in this commit:
[commit](https://github.com/IRNAS/grbl_stm32/commit/b5362ff4210b91ef37b57005d5482818384f8c07)
:smile:

Andypos, 2016-04-05:
@VojislavM  Good job :). I do mechanical part ,now. I bulding H-bot ( work like corexy) version.
If You change in Arduino CNC sheffild logic for 3.3 V?

VojislavM, 2016-04-06:
@Andypos I am not sure what is your question, but I have no problem with [Arduino shield](http://blog.protoneer.co.nz/arduino-cnc-shield/) and Nucleo 3.3 V logic.

Andypos, 2016-04-06:
It is ok ,if do not use analog input. See [https://developer.mbed.org/questions/4738/5V-tolerant-pins-on-Nucleo-boards/](url)

J-Dunn, 2016-04-24:
What is the development environment for this code ?  Is it intended to build under Mbed, Arduino "IDE" , eclipse... ? 

a quick README.md would be helpful to give an overview or what it is and how to build it. 

Thx.