langwadt/grbl_stm32 #5: Other STMicro Motor driver board support?

jdccdevel, 2016-06-27:
What other STM32 expansion boards than the IHM01A1 can I use with this code?

If that's the only board supported right now, how hard would it be to add support for a new board?

In particular, I'm looking into using the High-Power STMicro controller board, model [IHM03A1](http://www.st.com/content/st_com/en/products/ecosystems/stm32-open-development-environment/stm32-nucleo-expansion-boards/stm32-ode-move-actuate-hw/x-nucleo-ihm03a1.html) to drive some NEMA34 steppers for a 3 axis CNC router I'm looking to build.

I'm also interested in support for: 
- The 2 Axis Stepper controller, model [IHM02A1](http://www.st.com/content/st_com/en/products/ecosystems/stm32-open-development-environment/stm32-nucleo-expansion-boards/stm32-ode-move-actuate-hw/x-nucleo-ihm02a1.html) 
- The 3 phase brushless controller, model [IHM07M1](http://www.st.com/content/st_com/en/products/ecosystems/stm32-open-development-environment/stm32-nucleo-expansion-boards/stm32-ode-move-actuate-hw/x-nucleo-ihm07m1.html)

Thanks!

langwadt, 2016-06-28:
IHM03A1 in current mode shouldn't be a problem I've got it running already but haven't tested much yet.
IHM02A1 won't work, the voltage mode drive on works when using the motion engine not the step clock.
IHM07M1 is a whole different project on it's own because it would need encoders for feedback and a servo loop for each motor on top of the commutation needed for brushless

jdccdevel, 2016-06-28:
It's good to know that the IHM03A1 is working, that is the main one I was thinking of using.

I was thinking of using the IHM07M1 to control a spindle, with an encoder and some sort of feedback loop to keep a consitant rpm under different loads. (So no servo loop I believe...)  I have some coding experience, but I'm not familair with this code at all. How difficult do you think it would be to add support for that application?  If you think it's not too bad, I could probably tackle it.

I was also wondering about support for more than 3 of the STMicro boards. I've heard it's possible to hack support for 4 or more, do you have any experience with that? 

If supporting 4 or more expansion cards is out, I could probably work a standalone speed controller with a second STMicro32 or something, but that wouldn't be ideal.

langwadt, 2016-06-28:
IHM07M1 is too wimpy for a spindle you'd need something more like IHM08M1 I'm not sure how much cpu it would take but unless you need to build a lot the much easier and cheaper solution is to use a standard brushless ESC it only needs power and pwm control, that's what I use. A connection to one of the motor wires should be enough to sense the speed and close the loop

I use a stack of 16 IHM03A1 for a different project not with GRBL though, you need to bend legs to stack more than three but you need to do that anyway to get access to the step clock on more than one IHM03A1. As you start stacking the SPI transfer needed to change direction takes longer time  

I don't know if it is possible to stack different types of boards, but I doubt it unless they are very similar like IHM03A1 and IHM01A1

J-Dunn, 2016-09-24:
langwadt, could you point me to the sort of thing you are referring to ?  A quick search seems to be dominated by hobby RC market for drones !

What sort of units are good for CNC spindle? I have not got as far as worrying about the spindle yet, but looking ahead is best.

langwadt, 2016-09-24:
This is the general idea: https://youtu.be/7imn1-8KmEM driven with an ESC like this: http://www.hobbyking.com/hobbyking/store/catalog/101590_1_High.jpg

for anything serious you have to fabricate something to add some proper bearings like this guy
https://youtu.be/kWAZ5mPhZe8