langwadt/grbl_stm32 #8: Serial Connection Garbage Data

nsiatras, 2016-07-25:
Hello,

I imported the code to a new project, using the System Workbench,targeted to Nucleo -F401RE.
The code builds perfectly fine but as I try to connect to the serial com port I get garbage data. I am thinking that this is something that has to do with the clock (HSE_VALUE, HSI_VALUE etc..)

Do you have any idea on what this error might  be ?

Thank you

langwadt, 2016-07-25:
It is probably related to the clock and baud rate setup, but I don't have a board with a 401 so I can't easily debug it

nsiatras, 2016-07-26:
Do you use an external oscilator on your board? I dont...

langwadt, 2016-07-26:
I don't have an external oscillator, I use the 8MHz from the debugger. 
try changing the line 

#define HSE_VALUE    ((uint32_t)25000000)
to 
#define HSE_VALUE    ((uint32_t)8000000)

in the file CMSIS/device/stm32f4xx.h

nsiatras, 2016-07-26:
I tried it but still garbage data comes out from the serial...

langwadt, 2016-07-26:
did you define the compiler settings to 401 instead of 411?

closed 2016-07-29 by nsiatras