langwadt/grbl_stm32 #9: how to compile your project using keil mdk

shamsiqbal, 2016-08-25:
i am trying to compile your code in keil mdk uVision 5 but have no success yet can you plz help me out hoe to solve this issue i got flowing error list
`*** Using Compiler 'V5.06 update 1 (build 61)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Target 1'
linking...
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by gcode.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by limits.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by main.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by motion_control.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by nuts_bolts.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by planner.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by print.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by probe.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by protocol.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by report.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by serial.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by settings.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by spindle_control.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by stepper.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_probe_state multiply defined (by system.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by gcode.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by limits.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by main.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by motion_control.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by nuts_bolts.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by planner.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by print.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by probe.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by protocol.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by report.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by serial.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by settings.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by spindle_control.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by stepper.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_state multiply defined (by system.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by gcode.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by limits.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by main.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by motion_control.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by nuts_bolts.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by planner.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by print.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by probe.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by protocol.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by report.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by serial.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by settings.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by spindle_control.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by stepper.o and coolant_control.o).
.\Objects\grbl.axf: Error: L6200E: Symbol sys_rt_exec_alarm multiply defined (by system.o and coolant_control.o).
Not enough information to list image symbols.
Not enough information to list the image map.
Finished: 2 information, 0 warning and 45 error messages.
".\Objects\grbl.axf" - 45 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:04`

can you plz make keil project port of your existing code 
regards

langwadt, 2016-08-27:
I don't have Keil so there isn't much I can do. Why do you need to use keil instead of openstm32 ?

shamsiqbal, 2016-08-28:
i want to build stand alone control while using stm32f429i discovery kit also want to add some more features in the code and for graphics things its easier to use keil mdk vs other platform its since more than a month i am not able to compile this grbl port on keil mdk any how thanks for reply