langwadt/grbl_stm32 #10: Compile using online mbed for F401re board

jappiemike, 2016-12-10:
Hello, I'm just starting off making a router/cnc and have some Nucleo401re boards lying around . Can I compile your source code with the online compiler for mbed and are there any issues that you know of compiling this for the F401re board (or any other mbed compatible board)? 
Thanks 
Mike

langwadt, 2016-12-10:
never used mbed so I don't know, seems like a silly toy that doesn't even support any kind of real debugging. as long as you are using STM32 why not download and install 
http://www.st.com/en/development-tools/sw4stm32.html

jappiemike, 2016-12-10:
I'm not really a programming junkie, my interest is really in getting your grbl code to work on a F401RE, that's why I'm using the Mbed online compiler. I'll give it a try and see what happens and keep you posted.