langwadt/grbl_stm32 #11: 3-Motor Configuration

MikePearson, 2017-02-21:
Hello,
What is the correct binary file for 3-Motor configuration? It seems there's supposed to be a link but there isn't one. I'm using a NUCLEO-F411RE and 3 X-NUCLEO-IHM01A1 boards. I'm assuming I can stack these? I've configured the jumpers accordingly. Everything seem to be working but the motors don't engage and therefore don't move using grbl Panel.

Thanks!

MikePearson, 2017-02-22:
My main issue is that the link to the binaries isn't actually a link. See attachment. Would you mind providing the binary for this configuration?

[image]

Thank you

langwadt, 2017-02-22:
The documentation that mentions a binary wasn't done by me, it was done by someone who used my port of the code. 

I can make you a binary for a stack with 3 ST drivers but the stepper current is build to the code so you will have tell me what current for each of the axis

MikePearson, 2017-02-22:
That would be very generous of you and much appreciated. My motors are rated at 2.8A for each axis.

langwadt, 2017-02-23:
try the new bin file: https://github.com/langwadt/grbl_stm32/blob/master/Debug/grbl_stm32.bin 

I expect you will probably need to use a fan when running at 2.8A

closed 2017-02-23 by langwadt

MikePearson, 2017-02-24:
> try the new bin file: https://github.com/langwadt/
> grbl_stm32/blob/master/Debug/grbl_stm32.bin
>
> I expect you will probably need to use a fan when running at 2.8A

We have moving motors. Thank you very much. I agree a fan will be a good
Idea. Based on your motor current comment from yesterday I
found cpu_map_stm32f411_nucleos.h. Is that the only file that needs to be
modified for this configuration? I realize that more than just the current
parameters need tweaking. If the source for this configuration was
available I'd be interested, but it isn't a bit deal. You've provided
enough.

Thanks again!

langwadt, 2017-02-24:
@MikePearson Yes the only file that needs changing should be cpu_map_stm32f411_nucleos.h

just comment out "#define STANDARD_GRBL" to get the 3xL6474 configuration

MikePearson, 2017-02-24:
Perfect!

gottmoz, 2017-02-25:
Hello!
I tried this file with stm32f411 and 3xl6474 with jumpers set, ut only the top driver is enabled...

langwadt, 2017-02-25:
@gottmoz you must have a problem in the hardware