langwadt/grbl_stm32 #14: PMW trouble.

J-Dunn, 2019-04-19:
Hi,

I want set up an opto isolated 10V control signal for my chinese VFD for the spindle. Having set up the opto I went to test and found I don't have a PWM pulse train on PB0 CN7/34 ( Arduino A3 ). 

Am I correct in thinking I just need to  $X  then M3 S500 to get a pulse train on that pin? 

thanks. I can't see what I'm missing here.

langwadt, 2019-04-20:
the STANDARD_GRBL define puts the spindle pwm on PA6

J-Dunn, 2019-04-20:
Thanks for the reply. That was the wrong code snippet actually , not sure what  happened. I'm not using STANDARD_GRBL . I'm using your original wiring on PB0 as I detailed. Anyway, I've got it working now and have the PWM driving the 0-10V spindle speed signal via an opto. Seems fine. 
I have 3 limits switches and z-probe similarly isolated.

I have a spare opto channel free which I'm trying to find a use for to pre-empt some future need before I box it all up and put it back on the machine.