langwadt/grbl_stm32 #15: Spindle spin-up pause

J-Dunn, 2019-08-07:
Hi, 

I notice in the code you put a "hack" in to allow a slow start of your spindle controller. 

I have a chinese VFD with PID controlled spin up and speed control, so in principal I should not need this.  I have a PWM on PB0 and it controls the spindle speed quite satisfactorily. 

However, I can not seem to get any delay from G4 commands. 

If I send  
_G4S5 
S10000M3Y-5_  

It moves off directly and does not seem to recognise the G4 pause. 

Is this what you were trying to work around or was that something else? 
Do you get a delay from  G4S5 ?

langwadt, 2019-08-07:
as far as I can tell G4 work just fine, what are you trying to do? if you want a delay after starting the spindle the G4 should be after the M3

J-Dunn, 2019-08-07:
I've just checked back on the output. It was not as I recalled. The initial move was done before I tried G4.  In fact any attempt at a G4 , with or without S5 get an error like "invalid command" :?

J-Dunn, 2019-08-07:
Jeez, just me being dumb and mixing stuff up.  G4p5 works as expected. 
Thanks for the confirmation, it's always helpful to have a check since we can't be to numerous using this port.  Apart from that little brain fart, it seems to be working great.

langwadt, 2019-08-08:
I don't even use it, I use a port of grbl-mega on a board I build with powerstep drivers